neuropsychology/NeuroKit.py #6: ECG: Localize T wave

DominiqueMakowski, 2017-04-26:
This would be useful for being able to characterize periods of systole (R-T interval) and diastole. More and more papers ([example](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5247580/)) suggest that the cardiac phase has an impact on the processing of a stimuli.

We could then add a column within the processed dataframe corresponding to an array of zeros and ones, called for example 'Systole' that would indicate the cardiac phase.

Found some [matlab code](http://fr.mathworks.com/help/signal/examples/peak-analysis.html) for the localization of QRS waves that, maybe, could be adapted?

@rhenanbartels do you think that detecting the t wave would be complicated?

DominiqueMakowski, 2017-04-28:
Looking at biosppy's R peak detector, I found [this line](https://github.com/PIA-Group/BioSPPy/blob/master/biosppy/signals/ecg.py#L983) that says: "if current slope is smaller than half the previous one, then it is a T-wave". 

However, the code's a bit to complex for understanding how to extract them.

DominiqueMakowski, 2017-04-28:
If we find a way to localize T waves, we could also localize the Q wave, then extract the QT intervals. This would be a nice feature for medical studies and assessments (the QT interval being an important feature of cardiac wellness).

rhenanbartels, 2017-04-28:
Maybe we can clone the repository and change the code to return  a list where the algorithm found a T-wave. I will try this here with a well-behaved ecg signal

DominiqueMakowski, 2017-05-02:
might be useful someday: http://ieeexplore.ieee.org/abstract/document/1275572/

DominiqueMakowski, 2017-05-19:
Done - Needs validation.
Note that QT interval length computation remains impossible as it uses the beginning of both waves, and I managed to get the peak location 😢

closed 2017-05-19 by DominiqueMakowski