neuropsychology/NeuroKit.py #9: numpy has no attribute 'norm'

stjep, 2017-06-07:
Trying to work through the [BioSignals processing in Python](http://neurokit.readthedocs.io/en/latest/tutorials/Bio.html) tutorial, but didn't make it far because I hit the above error when trying to run:

`bio = nk.bio_process(ecg=df["ECG"], rsp=df["RSP"], eda=df["EDA"])`

The full Traceback from the IPython console is:

`Traceback (most recent call last):

  File "<ipython-input-280-be922da6b29f>", line 1, in <module>
    bio = nk.bio_process(ecg=df["ECG"], rsp=df["RSP"], eda=df["EDA"])

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/neurokit/bio/bio_meta.py", line 112, in bio_process
    ecg = ecg_process(ecg=ecg, rsp=rsp, sampling_rate=sampling_rate, resampling_method=resampling_method, quality_model=ecg_quality_model, hrv_segment_length=hrv_segment_length, age=age, sex=sex, position=position)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/neurokit/bio/bio_ecg.py", line 208, in ecg_process
    processed_ecg["ECG"]["HRV"] = ecg_hrv(rri, sampling_rate, segment_length=hrv_segment_length)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/neurokit/bio/bio_ecg.py", line 665, in ecg_hrv
    hrv["Entropy_Multiscale"] = entropy_multiscale(rri, emb_dim=2)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/neurokit/signal/complexity.py", line 408, in entropy_multiscale
    se = nolds.sampen(temp_ts, 1, tolerance, dist="euler", debug_plot=False, plot_file=None)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/nolds/measures.py", line 653, in sampen
    dsts = np.norm(diff, axis=1)

AttributeError: module 'numpy' has no attribute 'norm'`

DominiqueMakowski, 2017-06-23:
@stjep, sorry for the delay of my answer, I was away 😄 

This error is, indeed a bit weird, and seems related to the nolds package. Have you try reinstalling the nolds package directly from its github repo? 

`pip uninstall nolds`
`pip install https://github.com/CSchoel/nolds/zipball/master/`

If it still don't work, I'll have a deeper look within `nolds` code 😉

DominiqueMakowski, 2017-06-27:
@stjep yesterday the creator of nolds pushed the last version on pypi. So now you can just directly uninstall and reinstall nolds from pypi and it should work. Let me know if it's the case so I can close this issue.

stjep, 2017-06-29:
I upgraded the nolds package and that has indeed fixed this particular. Running that same command gets me a new bug, for which I've created a new issue (I thought it'd be neater, and in case anyone else runs into the same issue).

closed 2017-06-29 by stjep