neuropsychology/NeuroKit.py #11: IndexError: arrays used as indices must be of integer (or boolean) type

Raphael-Rissler, 2017-06-27:
Hello there, 

I try to analyze ECG and EDA data with your gr8 tool. Unfortunately I ran into a problem. 
What I've done so far:
1. Downloaded working winpython distrib. as outlined here (http://neurokit.readthedocs.io/en/latest/tutorials/Python.html)
2.  Run the following Code (see attached file for review input data 
[Data.xlsx](https://github.com/neuropsychology/NeuroKit.py/files/1104982/Data.xlsx))
```
# Import packages
import neurokit as nk
import pandas as pd
import numpy as np
import biosppy as biosppy

# Plotting preferences
%matplotlib inline

import matplotlib
matplotlib.rcParams['figure.figsize'] = [14.0, 10.0]  # Bigger figures
import seaborn as sns
sns.set_style("whitegrid")  # White background
sns.set_palette(sns.color_palette("colorblind"))  # Better colours

df = pd.read_csv("C:/Users/.../Data.csv", sep=';')
df=df.rename(columns = {'Input2':'EDA'})
df=df.rename(columns = {'Input3':'ECG'})

bio = nk.bio_process(ecg=df["ECG"], eda=df["EDA"])
```

### However, execution ends with an error

```
bio = nk.bio_process(ecg=df["ECG"], eda=df["EDA"])
Traceback (most recent call last):

  File "<ipython-input-40-84efa8150f06>", line 1, in <module>
    bio = nk.bio_process(ecg=df["ECG"], eda=df["EDA"])

  File "C:\Users\...\Desktop\neuropython\WinPython-64bit-3.5.3.1Zero\python-3.5.3.amd64\lib\site-packages\neurokit\bio\bio_meta.py", line 95, in bio_process
    eda = eda_process(eda=eda, sampling_rate=sampling_rate, use_cvxEDA=use_cvxEDA)

  File "C:\Users\...\Desktop\neuropython\WinPython-64bit-3.5.3.1Zero\python-3.5.3.amd64\lib\site-packages\neurokit\bio\bio_eda.py", line 94, in eda_process
    scr_onsets[biosppy_eda['onsets']] = 1

IndexError: arrays used as indices must be of integer (or boolean) type
```

...With all other input data, it worked fine. Only this data-set seems to make problems but I cannot figure out why. Would be nice if you could help me to resolve this.

Thanks in advance

DominiqueMakowski, 2017-06-27:
@Raphael-Rissler, could you send me this file? You can either attach it here or send it to me (dom.makowski@gmail.com) if you don't want it to go public. I'll try to figure out what's going on 😉

DominiqueMakowski, 2017-06-27:
Sorry I did not see your file, I'll have a look this evening or tomorrow!

DominiqueMakowski, 2017-06-27:
@Raphael-Rissler The excel file you provided seems to work just fine (see attached picture).
Maybe something went wrong when you converted it to a csv. If the separators are ";", are you sure decimal delimiters are "." and not ","? If you do not succeed to fix it, could you send me the csv with which neurokit throws an error. Thanks!
[image]

Raphael-Rissler, 2017-07-31:
Hello, 

first sorry for my late response, was travling... :)
I doublechecked again but the error occured again in some other files. I've send you a CSV, which I cannot analyse unfortunately due to the error mentioned above.

[2_a02_Bioplux.zip](https://github.com/neuropsychology/NeuroKit.py/files/1186593/2_a02_Bioplux.zip)

DominiqueMakowski, 2017-08-07:
@Raphael-Rissler I will answer also your other issue (#27) here as I think the problem is related.

So, I did the following:

```python
import neurokit as nk
import pandas as pd

df = pd.read_csv("2_a02_Bioplux.csv", sep=";")

df["Input1"].plot()
df["Input2"].plot()
df["Input3"].plot()

"""
I believe that Input3 is the ECG channel (with reversed polarity) while Input2 looks like EDA signal. As for Input1, I've no idea what it is. Note that units of the ECG are unusual (not in mV).
"""

df["Input3"][0:10000].plot()

"""
I plotted a subset of 10,000 timepoints. As we can see, there's about 13 heart beats, which is plausible for a 10 seconds window. That suggests that the signal is sampled at 1000Hz.
"""

# Let's see if preprocessing works:
processed = nk.ecg_preprocess(ecg=df["Input3"])

rpeaks = processed["ECG"]["R_Peaks"]

nk.plot_events_in_signal(signal=df["Input3"][0:10000], events=rpeaks[0:13])  # the segmenter correctly identified the R peaks.
```

While the preprocessing works, the "ecg_process" function does not (*ie* takes much longer than it should). This might be caused, indeed, by the polarity reversal you mentionned in #27. However, many things (including HRV, RSA, heart rate, and stuff) rely on the R peaks location list. So if those are correct, than the computation should be correct. However, other indices might not be correct.

I suggest you to 1) indentify which features are of interest to you, 2) See in the code on what indices they rely ([here](https://github.com/neuropsychology/NeuroKit.py/blob/master/neurokit/bio/bio_ecg.py#L372) you can see, for example, that ecg_hrv only requires r_peaks) 3) If they rely on indices that were correctly computed by preprocessing, apply the required functions

Let me know how it goes 😉