neuropsychology/NeuroKit.py #12: Data must be 1-dimensional

stjep, 2017-06-29:
This is a follow-on from issue #9.

When following the steps in the [tutorial](http://neurokit.readthedocs.io/en/latest/tutorials/Bio.html), I'm still stuck on the ecg process line (`bio = nk.bio_process(ecg=df["ECG"], rsp=df["RSP"], eda=df["EDA"])`).

Full traceback:

```
/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator LabelBinarizer from version 0.18.1 when using version 0.18.2. This might lead to breaking code or invalid results. Use at your own risk.
  UserWarning)
/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator MLPClassifier from version 0.18.1 when using version 0.18.2. This might lead to breaking code or invalid results. Use at your own risk.
  UserWarning)
Traceback (most recent call last):

  File "<ipython-input-8-be922da6b29f>", line 1, in <module>
    bio = nk.bio_process(ecg=df["ECG"], rsp=df["RSP"], eda=df["EDA"])

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/neurokit/bio/bio_meta.py", line 112, in bio_process
    ecg = ecg_process(ecg=ecg, rsp=rsp, sampling_rate=sampling_rate, resampling_method=resampling_method, quality_model=ecg_quality_model, hrv_segment_length=hrv_segment_length, age=age, sex=sex, position=position)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/neurokit/bio/bio_ecg.py", line 208, in ecg_process
    processed_ecg["ECG"]["HRV"] = ecg_hrv(rri, sampling_rate, segment_length=hrv_segment_length)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/neurokit/bio/bio_ecg.py", line 671, in ecg_hrv
    hrv["Lyapunov"] = np.max(nolds.lyap_e(rri, emb_dim=58, matrix_dim=4))

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/nolds/measures.py", line 514, in lyap_e
    mat_X -= data[i:i + emb_dim:m]

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pandas/core/ops.py", line 727, in wrapper
    dtype=dtype,

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pandas/core/ops.py", line 635, in _construct_result
    return left._constructor(result, index=index, name=name, dtype=dtype)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pandas/core/series.py", line 248, in __init__
    raise_cast_failure=True)

  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pandas/core/series.py", line 3027, in _sanitize_array
    raise Exception('Data must be 1-dimensional')

Exception: Data must be 1-dimensional
```

DominiqueMakowski, 2017-06-29:
@stjep I've just tried it with the minimal following code and the `bio_process()` works at my place. 

```python
import neurokit as nk
import pandas as pd

df = pd.read_csv("https://raw.githubusercontent.com/neuropsychology/NeuroKit.py/master/examples/Bio/data/data_bio.csv", index_col=False)  # download the datafile directly from the web
bio = nk.bio_process(ecg=df["ECG"], rsp=df["RSP"], eda=df["EDA"])  # run the processing function
```
Here's a few things:
- The biosignals tutorial is somehow out-dated (I've made several (altough not *that* major) changes to the API (mainly variable names and stuff) and I didn't have the time to refactor the tutorial. I'm planning to do it within the next weeks or days). Meanwhile, do not hesitate to discuss every issues here (even your tinier interrogations or remarks 😉, I'll be happy to help.
- With that being said, here are some thoughts about  why it doesn't work on your computer:
  - The error comes from this line ` hrv["Lyapunov"] = np.max(nolds.lyap_e(rri, emb_dim=58, matrix_dim=4))` within the ecg_hrv function.
  - However, as you can see in the [current code](https://github.com/neuropsychology/NeuroKit.py/blob/master/neurokit/bio/bio_ecg.py#L646), this line already created some troubles so that I recently put it within a `try` section to catch the error.
  - It is very likely that you missed this update I made few days ago (and that's my fault, I didn't even change the version number)
  - I suggest you reinstall neurokit: `pip uninstall neurokit` and then `pip install https://github.com/neuropsychology/NeuroKit.py/zipball/master`
  - It should work.
- However, note that, especially if you're interested in HRV (and the frequency domain features), this function is under [refactoring](https://github.com/neuropsychology/NeuroKit.py/issues/10) and should be finished this week (I hope). I'll push a new version then.

Thanks for your feedback,
Let me know how it goes!

stjep, 2017-06-29:
Your are correct that I did miss the update, and installing the current master fixed this issue.

I'm just getting my feet wet with ECG, so I really appreciate your quick and extremely helpful replies. I'll keep slowly playing with NeuroKit, and I'm tracking the repo so Github should (theoretically) notify me when new versions are pushed.

Seeing as how I'm staring out new with NeuroKit, let me know if there's any way I can contribute to the documentation.

DominiqueMakowski, 2017-07-05:
@stjep How's your biosginals exploration going? Was neurokit helpful? 
I've managed to rework the [tutorial](http://neurokit.readthedocs.io/en/latest/tutorials/Bio.html) a bit, let me know if you'd like to see more stuff 😉

closed 2017-07-05 by DominiqueMakowski

stjep, 2017-07-06:
@DominiqueMakowski, made my way through the tutorial without any problems, and am about to give my own data a try 🤞