neuropsychology/NeuroKit.py #26: Frequency output

AnteriorInsula, 2017-07-21:
Hi!  Sorry to keep asking questions - I think once I get more familiar with the NeuroKit/BioSPPy environment, I'll be able to start figuring them out on my own more.  But for now...

I'm trying to get the power in different frequency bands of HRV.  In the ECG output, I see columns labeled ECG_HRV_HF, ECG_HRV_LF, etc., with a value for each index.  I assume these correspond to high frequency, low frequency, etc., but I'm not sure how to extract the power in that specific frequency band, because I'm not sure what those values represent.  Do I need to get the area under the function that is described by those values?

Thanks for everything, you have been so helpful!

DominiqueMakowski, 2017-07-21:
Alright, so for this part, I'm not sure about what I'm saying, since I'm no expert of HRV nor signal processing. First of all, note that this part changed recently, so make sure you have version 0.1.8 installed. 
Then, you can find all the details about HRV computation [here](https://github.com/neuropsychology/NeuroKit.py/blob/master/neurokit/bio/bio_ecg.py#L569). Basically it computes power spectral density using the [multitaper](http://www.nmr.mgh.harvard.edu/mne/dev/generated/mne.time_frequency.psd_array_multitaper.html) method (*note to me to think about re-checking and implementing other methods*). Then it gets the area under the curve according to the standart frequency bands. But I am not 100% sure about that algorithm, nor the values it returns. It should be checked by an expert of the field (such as @rhenanbartels 😆)

AnteriorInsula, 2017-07-21:
Thanks Dominique!!!

closed 2017-08-07 by DominiqueMakowski