neuropsychology/NeuroKit.py #32: reading data from .acq

bobchicken, 2017-08-24:
Hi 

I'm trying to read my data from Acqknowledge file but it doesn't look right at all. It seems like it displays values for only one channel, where there are in fact three.  I just find nothing works on the Acqknowledge data - even the plots look wrong, displaying the signal for a few of the recorded minutes. I'm completely new to this so apologies if this is an obvious question - just completely lost on what to do with it. 

Thanks a lot

DominiqueMakowski, 2017-08-24:
@bobchicken Could you send me one acq file by mail at dom.mak19@gmail.com? I'll try to fix your problem

bobchicken, 2017-08-24:
Thanks a lot Dominique, files sent.

DominiqueMakowski, 2017-09-27:
@bobchicken did it work?

bobchicken, 2017-09-27:
Hi Dominique - sorry for the late reply I have been away from work for a
while! Thank you so much  for sending that over. It did work, however still
displaying only one channel. I'm going to work with a new Port next week so
will see how that goes.

Thanks a lot!

K

closed 2017-09-27 by DominiqueMakowski