neuropsychology/NeuroKit.py #36: HRV not defined

bbertoldi, 2017-09-26:
When trying to extract regular HRV indices, I am getting this error message:

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-8-619ff25e0114> in <module>()
----> 1 print(hrv.keys()) #

NameError: name 'hrv' is not defined

DominiqueMakowski, 2017-09-26:
@bbertoldi It depends on how you named the object returned by the `ecg_process()` function. Could you paste the code above that?

bbertoldi, 2017-09-26:
The only code I have above that is this one:

import neurokit as nk
 
data, sampling_rate = nk.read_acqknowledge("Zall_Baseline.acq", return_sampling_rate=True)
 
 
data[0:50000].plot()  # Plot a subset of raw data
 
processed_ecg = nk.ecg_process(ecg=data["ECG - ECG100C"], sampling_rate=sampling_rate)
nk.z_score(processed_ecg["df"][0:50000]).plot()
 
 
# Plot spectral HRV on a subset
nk.z_score(processed_ecg["df"][['ECG_Filtered', 'ECG_RR_Interval', 'ECG_HRV_HF', 'ECG_HRV_LF', 'ECG_HRV_ULF', 'ECG_HRV_VHF', 'ECG_HRV_VLF']][0:50000]).plot()
 
 
# Print HRV indices
print(processed_ecg["ECG"]["HRV"])

________________________________________________________
Since I'm completely new to this I'm still confused as to how to extract HRV indices from an Acknowledge file. Do I do this using the script above? How is the print(hrv.keys()) # and hrv["RMSSD"]
hrv["medianNN"] hrv["cvNN"] hrv["LF/HF"] different from what the above code does?

DominiqueMakowski, 2017-09-27:
@bbertoldi 

Here's a minimal code to check what HRV indices are available and extract (print) one of them.

```python
import neurokit as nk

# Preprocessing
data, sampling_rate = nk.read_acqknowledge("Zall_Baseline.acq", return_sampling_rate=True)
processed_ecg = nk.ecg_process(ecg=data["ECG - ECG100C"], sampling_rate=sampling_rate)

# What are the available indices?
print(processed_ecg["ECG"]["HRV"].keys())

# Print the RMSSD only
print(processed_ecg["ECG"]["HRV"]["RMSSD"])
```
Then again, in depends on what you want to do with these indices.

closed 2017-10-10 by DominiqueMakowski