neuropsychology/NeuroKit.py #37: Paper reference

Qiriro, 2017-10-21:
Thanks a lot for the great work that was put into implementing this library. I have been reading the source code related to the respiration signal processing (the [bio_rsp.py](https://github.com/Qiriro/NeuroKit.py/blob/master/neurokit/bio/bio_rsp.py)) but I could not fully understand how the algorithms were implemented. If possible, would you please provide a paper or any other reference this work was based on? Did you make any publication related to respiration signal?
Regards,
K

DominiqueMakowski, 2017-10-24:
Dear @Qiriro could you be a bit more precise; what function or lines are nebulous to you? I'll try to explain them as best as I can.

In general, NeuroKit is not published yet, and there are (to my knowledge) no papers using it for RSP (but the package is quite recent). Again, if you could tell me what function/lines you don't understand, I'll try to give you, if possible, some reference.

cheers

Qiriro, 2017-10-24:
Dear @DominiqueMakowski  thanks a lot for your reply and your assistance. I do understand the code. I, however, needed to know if there is any published paper that would give me more information about why you wrote the algorithms the way you did. For example, when calculating the respiration cycles and other respiratory related changes, did you use any published algorithms or it is completely based on your own (unpublished) algorithms? I needed this information to help me fully grasp why you made the decision you made. Otherwise, I fully understand your code and I thank you for kindly sharing.

DominiqueMakowski, 2017-10-24:
Unfortunately no, it is mostly based on my own unpublished algorithms. However, respiration cycles are quite straightforward as they correspond to the "raise" part of the signal (inspiration) and the descending part for expiration. 

Feel free to run some validation tests on some data and compare it with whatever other softwares (and if you do that, please let us know of the results :), we could add it in the documentation)

good luck!

closed 2017-10-24 by DominiqueMakowski