neuropsychology/NeuroKit.py #38: Number of peaks

Eichhof, 2017-11-07:
Hi,

The EDA stuff works but unfortunately I’m getting only peaks for 50% of my stimulus (10s pictures), i.e. in 50% of these 10s windows (picture duration) there is no peak. I have to evaluate a bit more to see the reason behind it (it seems a bit strange to me). Do you have any idea why this happens?

DominiqueMakowski, 2017-11-08:
It's indeed normal (and almost reassuring) to not have peaks for all of your stimuli, as the algorithm that finds the peaks is a bit more complex than only "taking the max value of a particular window". To be considered a true SCR peak, a peak must followed a certain pattern (see the [eda_scr](https://github.com/neuropsychology/NeuroKit.py/blob/master/neurokit/bio/bio_eda.py#L328) function for more details).

Following that, you have several options:
1) Do not bothering with the peaks and taking the max value of your EDA signal in a window
2) Using the *magnitude* of the peak, *ie* input 0 when there is no peak
3) Using the *amplitude* of the peak, *ie* the value of "true" (detected) peaks only
4) Using the number of peaks rather than their magnitude (= 80% of my emotional stimuli were followed by a peak, while only 20% of the neutral stimuli elicited a peak)
5) Decreasing the value of the `scr_treshold` parameter in `eda_process`. This will lower the treshold of peak detection and, thus, detect more "peaks"

Eichhof, 2017-11-08:
Thank you very much for your answer.

Regarding the first option, would you recommend taking the max of the raw EDA signal or the max of the phasic component?

DominiqueMakowski, 2017-11-08:
@Eichhof I would use the phasic component, especially if you're interested in emotional processing. However, it mostly depends on your needs as always 😸

Eichhof, 2017-11-08:
Thank you for the fast answer.

closed 2017-11-08 by Eichhof