neuropsychology/NeuroKit.py #40: Sampling rate for ECG feature extraction

Eichhof, 2017-11-07:
Hi,

I have a series of RR-intervals (which I have retrieved from Polar H10 chest strap)

The ecg_hrv function from neurokit, that computes HRV features, needs R peaks location as input. The step, in theory, before RRi computation. So we have to somehow transform the RRi data back into R peaks location. We can do it by computing the cumulative sum ` rpeaks = list(np.cumsum(rri)) `.

`rri_results = nk.ecg_hrv(rpeaks, sampling_rate=1000, hrv_features=["time", "nonlinear"])`
 
On this line you have used a sampling rate of 1000. What should I use here? Somehow the sampling rate depends on the heart rate (the higher the heart rate, the more RR intervals).

DominiqueMakowski, 2017-11-08:
The sampling rate refers, here, to the unit in which your R peaks location are expressed. 

For example, if you have a list of R peaks location such as `[5, 6.2, 7.85, 9.01, ...]`, it is likely to be expressed in seconds (sampling_rate=1). While if you had `[5000, 6200, 7850, 9010, ...]` it is rather a sampling_rate of 1000 points par second. 

Let me know if that remains unclear!

Eichhof, 2017-11-08:
Thank you for your answer. Now it is completely clear. ;)

closed 2017-11-08 by Eichhof