neuropsychology/NeuroKit.py #44: How to correctly doing baseline correction?

Eichhof, 2017-11-17:
Hi there

I have a quick question regarding baseline correction for HRV and EDA features. I'm extracting the following features (over a 10s window):

EDA: latency, rise time, number of peaks, mean and std of tonic and phasic
HRV: pnn20, rmssd, sdnn, sdss

I have also done a 5 minute baseline recording for each person at the beginning. I think there are several possibilities for baseline correction:

1. Dividing (or subtracting?) from each feature the minimum or mean value over the whole recording time (not the baseline).

2. Dividing (or subtracting?) from each feature the minimum or mean value over the whole baseline.

3. Dividing (or subtracting?) from each feature the minimum or mean value over e.g. the last 2 minutes of the baseline.

4. Dividing (or subtracting?) from each feature the minimum or mean value around a 30 seconds window where the heart rate or EDA was minimal in the baseline.

What would you propose, i.e. with which method did you have the best results?

Thanks a lot for the help!

DominiqueMakowski, 2017-11-17:
For Phasic EDA I'm not sure it actually makes sense to apply baseline correction. For HRV, I would rather correct for a window immediately preceeding your event. Then, the type of correction depends mostly on your hypotheses. 

For example, I presented emotionally negative pictures to participants. Based on the litterature, I expected a higher heart rate deceleration upon stimulus presentation (let's say, for 5 s). So I took the minimum of the heart rate in these 5 s and I subtracted that from the mean of a [-1, 0] window. To quantify the deceleration amplitude, so to speak. 

But again, I'm not sure there are any gold standarts, you should follow the litterature unless you've got a good (ie, justified) reason for doing something else 😄

Eichhof, 2017-11-17:
Thank you for your answer. What do you mean by the [-1,0] window? So you had a 5s window beginning at the stimulus onset and over this window you searched the minimum and subtracted this minimum from the mean over this window?

DominiqueMakowski, 2017-11-17:
@Eichhof I substracted the minimum of the 0-5s window of the mean of the 1-second-long window *preceeding* stimulus onset.

Eichhof, 2017-11-17:
Now I've got it. ;) Did you take the mean of the heart rate over 1 seconds (I don't know how that mean could be calculated over such a small period)?

DominiqueMakowski, 2017-11-17:
NeuroKit computes by default a third order spline interpolation of the RR intervals to get a continuous signal of heart rate. You could even take the value of the this heart rate signal upon stimulus onset. But otherwise, you can compute your own heart rate using R peaks (in that case you'll need indeed a larger window)

Eichhof, 2017-11-17:
Makes sense. :)

closed 2017-11-17 by Eichhof