neuropsychology/NeuroKit.py #46: Window size for EDA and HRV

Eichhof, 2017-11-19:
Hi there,

I'm using EDA and HRV features. I'm extracting these features for each stimulus by positioning a window at the start of the stimulus (I think including data before the stimulus does not make sense).

What window size is recommended for EDA and HRV? I've heard that for EDA the response should appear within 5 seconds.

DominiqueMakowski, 2017-11-20:
For EDA, in affective neuroscience, a window between 1 and 7 s or 1 and 5 should be ok to catch the response.

For HRV, there is no unique rule as there are like dozens of HRV indices. Again, a good start is to reproduce what was done in similar papers.

Eichhof, 2017-11-20:
I have taken the usual features according to literature. HRV featues work much better than EDA featues (perhaps due to the sensor, I don't know). I also has to throw out a lot of the featues due to high correlation (especially for HRV).

Currently, I'm using a 10s window. When I shrink the window to 5s, accuracy drops to random level. But I still have to evaluate it separately for HRV and EDA.

Regarding the frequency domain of HRV, I'm always getting 0 for the low and very low frequency (only high frequency exists). Perhaps this is dur to the small window size of 10 sevonds.

DominiqueMakowski, 2017-11-20:
yes, you can't estimate low or very low frequency domain HRV on 10 seconds. Well, try out a few things and find out what works better ;)