neuropsychology/NeuroKit.py #50: eda_scr errors

FabianHertwig, 2017-12-05:
[gsr_slice.p.zip](https://github.com/neuropsychology/NeuroKit.py/files/1531215/gsr_slice.p.zip)
Hi,

I'm trying to extract Skin Conductance Responses from a EDA Signal, but I'm getting errors.

Here is some example code and the errors and I'm attaching the file.

```(python)
import pickle
import neurokit as nk

gsr_slice = pickle.load(open("gsr_slice.p", "rb"))
nk.eda_scr(gsr_slice, 512, method="fast")

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-101-9f5b83c41189> in <module>()
----> 1 nk.eda_scr(gsr_slice, 512, method="fast")

~/anaconda/envs/kohaf/lib/python3.5/site-packages/neurokit/bio/bio_eda.py in eda_scr(signal, sampling_rate, treshold, method)
    408     # ====================
    409     # Compute rising times
--> 410     risingtimes = peaks-onsets
    411     risingtimes = risingtimes/sampling_rate*1000
    412 

ValueError: operands could not be broadcast together with shapes (1319,) (1309,) 

nk.eda_scr(gsr_slice, 512, method="slow")

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-103-cf6859b8848d> in <module>()
----> 1 nk.eda_scr(gsr_slice, 512, method="slow")

~/anaconda/envs/kohaf/lib/python3.5/site-packages/neurokit/bio/bio_eda.py in eda_scr(signal, sampling_rate, treshold, method)
    402 
    403     # Keep only pairs
--> 404     peaks = peaks[peaks > onsets[0]]
    405     onsets = onsets[onsets < peaks[-1]]
    406 

TypeError: only integer scalar arrays can be converted to a scalar index

```

DominiqueMakowski, 2017-12-05:
@FabianHertwig 

Neurokit works better with pandas. Also, you should first process your signal before extracting features.

Try the following:

```python
import pickle
import neurokit as nk
import pandas as pd

gsr_raw = pickle.load(open("gsr_slice.p", "rb"))
gsr_raw = pd.Series(gsr_raw)  # Convert to pandas series

gsr_processed = nk.eda_process(gsr_raw, sampling_rate=1000)  # Process the signal. Adjust the sampling rate correctly!!!

# Extract stuff
gsr_signal = gsr_processed["df"]
gsr_features = gsr_processed["EDA"]
gsr_peaks = gsr_features["SCR_Peaks_Indexes"]

# Plot peaks into signal
nk.plot_events_in_signal(gsr_signal, gsr_peaks)
```
[image]

FabianHertwig, 2017-12-05:
Thank you. The example helps a lot. This should be in the docs somewhere!

Processing is rather slow. What is the best method to speed it up? Downsampling? I read somewhere that a 10hz samplerate should be sufficient. My signal has a samplerate of 512 hz.

DominiqueMakowski, 2017-12-05:
It is indeed a slow process, and downsampling can speed it up a bit. 10 Hz should in theory be sufficient, but I'd rather go for 50 or 100 (personal opinion tho).

Glad it helped, good luck 😉

closed 2017-12-05 by DominiqueMakowski