neuropsychology/NeuroKit.py #61: NeuroKit Warning: ecg_hrv(): Correlation Dimension. Error:  NeuroKit warning: complexity_entropy_multiscale(): Signal might be to short to compute SampEn for scale factors > 0. Setting max_scale_factor to 0.

Sanjay1995, 2018-04-14:
I am giving the data in form of 1D numpy array which is basically i, ii, iii, v1, v2, v3, v4, v5, v6, avr, avl, avf features from PTB database. I call this function nk.bio_process(ecg=ecg_signal[:,i],ecg_quality_model=None). It operates good on several columns but gives error after 2 iterations. Moreover, I have checked data is fine enough in all columns. can anyone please resolve the issue

DominiqueMakowski, 2018-04-14:
Hi @Sanjay1995, could you provide an example of your dataset? We'll try to fix the error. It might be related to a recent change in HRV computation (#58). Also linking @gattia just in case :)

Sanjay1995, 2018-04-14:
@DominiqueMakowski

gattia, 2018-04-14:
Based on the print out it seems that the sample entropy is having problems on the very first pass at the full resolution scale for the multi-scale analysis (this should be the same as just running sample entropy on the full data). This shouldn't be a problem from anything done to the multiscale entropy function recently.  

If I were debugging it, I'd be interested in what the data that is being passed to `complexity_entropy_multiscale()` looks like - whats it's shape, min, max values, how does it looked graphed out, etc.

Sanjay1995, 2018-04-14:
[image]

Sanjay1995, 2018-04-14:
this is how my one data column which I (i) as I mentioned above looks like

Sanjay1995, 2018-04-14:
@gattia

Sanjay1995, 2018-04-14:
thanks @DominiqueMakowski it is really useful, but tell me ecg_preprocess() and bio_process() works alike in my case?

DominiqueMakowski, 2018-04-14:
@Sanjay1995  yes bio_process is just a wrapper for processing multiple signals (ECG, EDA, EMG etc.) at once. Using bio-process with only ecg is similar to using `ecg_process`. However, `ecg_process` uses itself, the `ecg_preprocess` function that only does low level preprocessing (mainly extracting R peaks and not computing more complex indices such as HRV for example).

Sanjay1995, 2018-04-14:
as I have already mentioned you that I am using dataset of PTB ecg dataset, and your library also inculcates it. But ecg_preprocess() also fails on some signals giving error (index 0 is out of bounds for axis 0) don't know why.

Sanjay1995, 2018-04-14:
If I only process LEAD 1 (i in my data), then would it give features which ('T_Waves', 'Cardiac_Cycles', 'P_Waves', 'Q_Waves', 'HRV', 'R_Peaks') helps me in classification of heart disease class.

DominiqueMakowski, 2018-04-14:
yes, you should use only the column of the dataset corresponding to `i`. I used the full PTB dataset only to create a machine learning model that automatically classifies the provided lead signal and returns the probability of correct classification (a proxy of [signal quality](http://neurokit.readthedocs.io/en/latest/features/ecg_quality.html)). But for investigating ECG features using only LEAD I is sufficient.

Sanjay1995, 2018-04-14:
@DominiqueMakowski  thanks.

waleedkaimkhani, 2018-04-14:
@DominiqueMakowski I am getting same error ecg_process index 0 is out of bound error but I have the signal of length greater than 1

DominiqueMakowski, 2018-04-14:
@waleedkaimkhani could you provide a sample of your data? thanks

waleedkaimkhani, 2018-04-14:
my dataset is ptb ecg database. you can check it on https://www.physionet.org/physiobank/database/ptbdb/

DominiqueMakowski, 2018-04-14:
haha alright;

1) did you select correctly a **one dimensional** array OR **one** pandas' dataframe column (corresponding to LEAD 1)? 
2) if yes, could you save (in txt, csv or json) this unique column or array and attach it here so I can check directly with the exact input you provide to neurokit's routines? Thanks 😅

waleedkaimkhani, 2018-04-14:
[image]

DominiqueMakowski, 2018-04-14:
or send it to me dom.makowski@gmail.com

waleedkaimkhani, 2018-04-14:
@DominiqueMakowski   i have sent you mail

DominiqueMakowski, 2018-04-17:
@waleedkaimkhani your code should look like that

```python
import neurokit as nk
import pandas as pd

df = pd.read_csv("file.csv")
ecg_processed = nk.ecg_process(ecg=df["i"], sampling_rate=1000)
```