neuropsychology/NeuroKit.py #64: Filtering for EDA

Eichhof, 2018-04-26:
Hi

Before applying the cvxEDA algorithm you are filtering the signal. According to the cvxEDA paper this is not necessary: "As a result, the method does not require pre-processing of the observed SC signal (e.g., bandpass filtering) nor post-processing of the inferred phasic and tonic components (e.g., to deal with negative neural activations)."

DominiqueMakowski, 2018-04-26:
@Eichhof as you can turn off the cvxEDA algorithm, a bit of filtering is indeed done by default. I had some good results with it (mainly to clean some serious artifacts to ease the subsequent cvxEDA computation). 

However, if you have reasons to think this could be a bad thing (the cvxEDA *must* be run on raw data), I believe we could either change it (and add the option to select between "traditional" filtering and cvxEDA) or a parameter to turn off the prefiltering.

What are your thoughts?

Eichhof, 2018-04-26:
Actually, I have just seen that in eda_process() you pass the raw (unfiltered and unsmoothed) signal to cvxEDA.