neuropsychology/NeuroKit.py #65: Baseline correcting signal

Eichhof, 2018-05-01:
Hello

I have a 7 minute nature video recording at the beginning of my experiment which I would like to use as baseline for my EDA and RR signal. Currently I'm just using the whole 7 minutes or the last 3 minutes which can be a bad choice because it is possible that a participant relaxes at the beginning but after some minutes starts getting annoyed of the video. I thought about searching the minimum EDA or RR/HR signal of the video baseline and then using for example a 10s window around this minimum as baseline.

Do you know other common possibilities? I did not find very much useful information about baseline correction for this special case.

Eichhof, 2018-05-01:
Hi Dominique

Thank you very much for your fast answer, it is great. In detail, I'm using the following features:

EDA: Latency, rise time, recovery time, number of peaks, mean/max/min/std of phasic and tonic signal, mean/max/min of amplitude

RR: mean, sdnn, sdss, pnn50, pnn20 as well as high and low frequency features

Regarding baseline correction I thought about calculating the above features in a 10 second window around the minimum of EDA or RR during baseline period. 

Which of the features would you baseline correct? I don't understand completely why amplitudes should not be baseline corrected. When I run cvxEDA then I still get differences in amplitudes between participants.

Eichhof, 2018-05-05:
Do you have any thoughts on this?

DominiqueMakowski, 2018-05-07:
@Eichhof 

For RRi, it often makes sense to correct for baseline as you're interested in RRi variations rather than absolute values. For example, let's say that the mean or minimum heart rate, during your experimental stimulus, is 80bpm. This could be a deceleration or a acceleration, depending on the baseline heart rate (if it was 100 or 60 bpm just before the stimulus).

To simplify, for EDA, the phasic component is detrended to keep only the SCRs. Other parts of the signal when there are no SCRs (which are event related responses) are 0. So you don't need to correct for the baseline, which is 0. Moreover, the values make sense in themselves:

> "Although individuals and experimental situations vary considerably, some very general estimations of 'typical' strength values can be provided. In terms of phasic SCRs, amplitudes can typically range from threshold to a maximum of around 2-3uS (X = 0.30uS - 1.30uS: raw scores, non-normalized) and around 0.20 - 0.60uS on average when logarithmized. In experiments where highly aversive / threatening / fearful stimuli are used, this maximum response can increase to around 8uS (non-normalized), but this is rare. Estimations for the Frequency of NS-SCRs are difficult, but an average of 1-3 per/min (with around 10 per/min being possible during periods of quietness) has been suggested. High arousal is associated with around 20-25 per/min (Boucsein, 2012). For SCL, a range of 1-40uS has been mentioned (Venables & Christie, 1980) though averages are more likely between 2 -16uS. These numbers can be influenced by the size of electrodes and this should be noted in all scientific reporting" ([here](https://www.birmingham.ac.uk/Documents/college-les/psych/saal/guide-electrodermal-activity.pdf))

Eichhof, 2018-05-07:
Hi Dominique

Thanks, that matkes sense. But when I look at the mean of the phasic component rather than the max amplitude then it makes again sense to baseline correct, am I right? I think the same should hold for the tonic component because it is not bounded to 0.

DominiqueMakowski, 2018-05-08:
I don't know, for EDA traditional outcomes are either the number of SCRs or their amplitude (peak max). I've personally never heard of averaging the signal