neuropsychology/NeuroKit.py #66: Cubic interpolation for frequency domain

Eichhof, 2018-05-10:
Hello

In bio_ecg.ecg_hrv you interpolate the RR intervals to convert to continuous RR interval before extracting the frequency domain features. You are doing this using a third order (i.e. cubic) spline. I assume that the way you calculate the frequency features needs a continuous signal. Would it not also be possible to compute frequency features without interpolation?

However, why are you using this cubic interpolation instead of a linear interpolation? I think with a cubic interpolation the signal can be overestimated or underestimated, especially around the RR values.

DominiqueMakowski, 2018-05-11:
@Eichhof glad to see you're digging into the code :)

Cubic interpolation is the method described in the papers I read. I believe this has to do with the underlying physiological dynamics (the heart rate would change "smoothly" rather than linearly with abrupt changes). However, you might want to ask an HRV expert to be sure.

Eichhof, 2018-05-11:
Thank you for your answer. Do you have some of the papers at hand which you mentioned?

DominiqueMakowski, 2018-05-22:
@Eichhof maybe in one of those:

- Heart rate variability. (1996). Standards of measurement, physiological interpretation, and clinical use. Task Force of the European Society of Cardiology and the North American Society of Pacing and Electrophysiology. Eur Heart J, 17, 354-381.
- Smith, A. L., Owen, H., & Reynolds, K. J. (2013). Heart rate variability indices for very short-term (30 beat) analysis. Part 2: validation. Journal of clinical monitoring and computing, 27(5), 577-585.
- Lippman, N. E. A. L., Stein, K. M., & Lerman, B. B. (1994). Comparison of methods for removal of ectopy in measurement of heart rate variability. American Journal of Physiology-Heart and Circulatory Physiology, 267(1), H411-H418.
- Peltola, M. A. (2012). Role of editing of R–R intervals in the analysis of heart rate variability. Frontiers in physiology, 3.