neuropsychology/NeuroKit.py #68: Hardcoded sampling rate?

Fegalf, 2018-05-31:
https://github.com/neuropsychology/NeuroKit.py/blob/30710dba4200f18d75a9faf54ea0a13e749248fc/neurokit/bio/bio_ecg.py#L580

If I understand correctly, this is an hard coded value of 1000Hz for "sampling_rate" variable. There seems to be many other instances of this elsewhere in the code.

DominiqueMakowski, 2018-05-31:
@Fegalf I think that it is because it was previously rescaled to a specific rate (for rejecting outliers in "seconds") for example here:

https://github.com/neuropsychology/NeuroKit.py/blob/30710dba4200f18d75a9faf54ea0a13e749248fc/neurokit/bio/bio_ecg.py#L504

It is then rescaled to "milliseconds" to compute the frequency domains indices (which need the same scale).

It's also possible that it's a bug, in which case do not hesitate to report or correct it ;)

Fegalf, 2018-06-04:
@DominiqueMakowski  You were right! It's only a rescaling to milliseconds units. My bad :/ !

closed 2018-06-04 by Fegalf