neuropsychology/NeuroKit.py #72: AttributeError: module 'neurokit' has no attribute 'bio_process'

ursusaquarius, 2018-07-04:
I just wanted to try the features of your neurokit.
I´m using Anaconda installed on Windows 10.
The pip install of neurokit was successful.
Then I copied your following sample code into a Spyder window:
 
# Import packages
import neurokit as nk
import pandas as pd
import numpy as np
import seaborn as sns
# Download data
df = pd.read_csv("https://raw.githubusercontent.com/neuropsychology/NeuroKit.py/master/examples/Bio/bio_100Hz.csv")
# Plot it
df.plot()
# Process the signals
bio = nk.bio_process(ecg=df["ECG"], rsp=df["RSP"], eda=df["EDA"], add=df["Photosensor"], sampling_rate=100)
# Plot the processed dataframe, normalizing all variables for viewing purpose
nk.z_score(bio["df"]).plot()
pd.DataFrame(bio["ECG"]["Cardiac_Cycles"]).plot(legend=False)
 
The first part imcuding the first plot runs well until nk.bio_process where I get this error message:
bio = nk.bio_process(ecg=df["ECG"], rsp=df["RSP"], eda=df["EDA"], add=df["Photosensor"], sampling_rate=100)
AttributeError: module 'neurokit' has no attribute 'bio_process'
 
What went wrong ?

DominiqueMakowski, 2018-07-09:
@ursusaquarius Sorry for the delay, I've been away. 

This error is indeed weird, as it suggests that 1) neurokit is *detected* (otherwise it would throw an "there is no module called neurokit" but 2) there neurokit's functions are not available.

Could you run the following to check the version:

```python
import neurokit as nk
nk.__version__
```

thanks

ursusaquarius, 2018-07-11:
Hi Dominique,

running these lines created the following output in the console window:

import neurokit as nk
nk.__version__
Out[1]: '0.2.5'

Hope this helps to solve this issue I had.

Best regards
Reinhold

DominiqueMakowski, 2018-07-11:
@ursusaquarius Ok. 

Could you try running another function of the package? After you loaded the data, instead of running the bio_process function, try:

`bio = nk.ecg_process(ecg=df["ECG"], rsp=df["RSP"], sampling_rate=100)`

ursusaquarius, 2018-07-11:
now I ran this and also again the bio_process function without any error
and could also plot the results.
The only change I did meanwhile was updating Spyder ...
Funny and I don´t really understand ...
Any idea what happened from your side ?

ursusaquarius, 2018-07-11:
my project is mainly about ECG and what information could be extracted, starting with raw data received from an ADC.
Would you know if Python could handle this fast enough eg on a ESP32 ?
Best regards

Outlook für iOS<https://aka.ms/o0ukef> beziehen

No idea... But glad it works now :)

DominiqueMakowski, 2018-07-12:
@ursusaquarius Sorry, I've no experience with that kind of stuff 😕

hantekno, 2019-01-21:
I get same issue while using it in spyder (based on anaconda, in windows 10), module neurokit has no attribute bio_process. Even im trying to check its version, it still getting the same notification (no atribute __version__). Can you provide the solution for this issue? thanks dominique

DominiqueMakowski, 2019-01-21:
@hantekno, could you confirm that the command `import neurokit as nk` works?

hantekno, 2019-01-21:
@DominiqueMakowski yup, the command work, but still the attribute doesnt work. Actually, i tried it on jupyter notebook, and its work. If you're not busy, can i ask you some question via email? i'm a student, totally new in python and ecg, but i want to learn it, thank you so much

DominiqueMakowski, 2019-01-21:
@hantekno If your questions are not confidential, I would prefer you to open a [new issue](https://github.com/neuropsychology/NeuroKit.py/issues) so it can help others 😉

closed 2019-01-21 by DominiqueMakowski