neuropsychology/NeuroKit.py #75: Linear Trend in EDA

Eichhof, 2018-08-28:
Hey

Do you know if the cvxEDA algorithm already removes linear trends? I think linear trends should not be a problem for phasic components but for tonic components.

DominiqueMakowski, 2018-08-28:
@Eichhof It does implement some form of trend correction (I believe it's more complex than a simple linear detrend tho). This algorithm was built and tested on raw data.

Eichhof, 2018-08-28:
Thanks for your answer. So you mean cvxEDA or your library implements some form of trend correction?

DominiqueMakowski, 2018-08-28:
I mean that the cvxEDA algorithm (present in neurokit but mostly copied from [here](https://github.com/lciti/cvxEDA)) does itself everything necessary and can be used with raw data