neuropsychology/NeuroKit.py #76: compute_dprime function has n_hits, n_miss, n_cr, n_fa hard coded

boomsbloom, 2018-09-06:
Hi--Thanks for developing this repo. I wanted to compare my own dprime/aprime calculations with ones from other repos and found the following:

https://github.com/neuropsychology/NeuroKit.py/blob/867655f84bf210626649bca72258af6a2b5a2791/neurokit/statistics/routines.py#L81

Should be an easy fix. Thanks!

DominiqueMakowski, 2018-09-07:
Indeed a bug that must have passed through the tests! thanks for noticing!!
Should be fix on master now

boomsbloom, 2018-09-07:
No worries! It also looks like there's a bug in the dprime calculation where hit_rate_adjusted is subtracted from hit_rate_adjusted rather than fa_rate_adjusted

https://github.com/neuropsychology/NeuroKit.py/blob/867655f84bf210626649bca72258af6a2b5a2791/neurokit/statistics/routines.py#L97

DominiqueMakowski, 2018-09-07:
Damn! lucky you're here :)

boomsbloom, 2018-09-11:
hah, no problem! Gonna close this issue now.

closed 2018-09-11 by boomsbloom