neuropsychology/NeuroKit.py #77: NeuroKit outdated installation

SIHHUA, 2018-09-19:
Hello,I have some question,and need your help
I wanted to use Neuropython but it's still failed
I ran the code,and it made error

`error`
module 'biosppy.signals.ecg' has no attribute 'correct_rpeaks'

I checked the file 'biosppy.signals.ecg', there has function'correct_rpeaks',and already has attribute
Is file "bio_ecg_preprocessing" wrong or file "biosppy.signals.ecg"?

`code`

> import neurokit as nk
> import pandas as pd
> import numpy as np
> import seaborn as sns
> 
> df =  pd.read_csv("C:/Users/User/Desktop/CH1.csv")
> df.plot()
> 
> bio = nk.ecg_process(ecg=df["ECG1"], rsp=None, sampling_rate=250, filter_type='FIR', filter_band='bandpass', filter_frequency=[3, 45], segmenter='hamilton', quality_model='default', hrv_features=['time', 'frequency'], age=None, sex=None, position=None)
> 
> nk.z_score(bio["df"]).plot()
> 

I tried to unistall neurokit ,and upgrade it
but it still wrong>> C:\Users\User\AppData\Local\Temp\pip-j3oj0gb4-build
What can i do?

DominiqueMakowski, 2018-09-19:
Related to [this issue](https://github.com/PIA-Group/BioSPPy/issues/44).

Don't worry, we'll fix this.

First of all, I have some questions:

- Did you already use python for anything else?
- Does the python you downloaded work, besides neurokit-related stuff (*i.e.,* Can you run `import numpy as np` and `np.mean([1,2,3])` in Spyder?

DominiqueMakowski, 2018-09-20:
@SIHHUA I have updated the neuropython version, could you try downloading it and running to see if it works? Here's the [link](https://neurokit.readthedocs.io/en/latest/tutorials/Python.html) thanks!

SIHHUA, 2018-09-30:
@DominiqueMakowski 
Did you already use python for anything else?>>No
Does the python you downloaded work, besides neurokit-related stuff (i.e., Can you run import numpy as np and np.mean([1,2,3]) in Spyder? >>Yes，it printed 2.0,and I tested other code which could run.

DominiqueMakowski, 2018-10-01:
At what moment does this error appear? Could you provide some reproducible code / steps required to reproduce this error?

SIHHUA, 2018-10-01:
I opened WinPython Command Prompt.exe to install NeuroKit, as shown below

[image]

But it could install Neuropsydia successfully

DominiqueMakowski, 2018-10-02:
@SIHHUA That's strange indeed, what is your python version?

Could you try, from [here](https://neurokit.readthedocs.io/en/latest/tutorials/Python.html), installing the neuropython distribution (where neurokit is already installed)? Does this work?

SIHHUA, 2018-10-02:
@DominiqueMakowski 
Python version is Python 3.6.6
Yes,I downloaded Neuropython.zip
[image]
And followed hard steps
[image]
It didn't work,because neurokit can't download successfully >>  No module named 'neurokit'