neuropsychology/NeuroKit.py #78: TypeError when analysing EDA

jaros1024, 2018-09-25:
Hello, I have a problem when trying to analyse EDA. Here is the stacktrace:
`File "C:\Users\jaros1024\PycharmProjects\Project1\lib\new_gsr.py", line 13, in __init__
    self.data = nk.eda_scr(y, sampling_rate=freq)
  File "C:\Users\jaros1024\PycharmProjects\Project1\venv3\lib\site-packages\neurokit\bio\bio_eda.py", line 399, in eda_scr
    peaks, _ = biosppy.tools.find_extrema(signal=signal, mode='max')
  File "C:\Users\jaros1024\PycharmProjects\Project1\venv3\lib\site-packages\biosppy\signals\tools.py", line 1108, in find_extrema
    values = signal[extrema]
TypeError: only integer scalar arrays can be converted to a scalar index`

I tried Python 3.6 and 3.7, both with numpy 1.14.6 and 1.15.2. Same effect for all. As far as I know, something was changed in recent numpy versions, so the syntax that you used is no longer supported.
Could you track it down and try to implement a workaround for this issue?

DominiqueMakowski, 2018-09-25:
@jaros1024 That's weird indeed, I'll look at it :) 
Could you tell me what is your version of NeuroKit and Biosppy?

jaros1024, 2018-09-25:
I just noticed that NeuroKit version 0.2.0 installed, but just updated to 0.2.7 and the problem is still the same. Biosppy is 0.6.1 - so I haven't changed anything.

DominiqueMakowski, 2018-09-25:
Great, could you provide a sample of data and a reproducible code snippet so that I can reproduce it and dig into it? Thanks!

jaros1024, 2018-09-26:
Sure! See the attached file. It's just a pickle dump, so you can open it like:
`import pickle

def open_file(in_file):
    with open(in_file, 'rb') as fp:
        itemlist = pickle.load(fp)
        return itemlist`
[sample.zip](https://github.com/neuropsychology/NeuroKit.py/files/2421413/sample.zip)

DominiqueMakowski, 2018-09-27:
It works for me with this code:

```python
import pickle
import neurokit as nk
import pandas as pd

def open_file(in_file):
    with open(in_file, 'rb') as fp:
        itemlist = pickle.load(fp)
        return itemlist

data = open_file("sample.dat")
data = pd.DataFrame({"EDA" : data})  # Convert to dataframe
data.plot()  # Check signal
```
[image]
```python
results = nk.eda_process(data["EDA"], sampling_rate=100)

# Plot standardized data (visually more clear) wiht SCR peaks markers
nk.plot_events_in_signal(nk.z_score(results["df"]), results["EDA"]["SCR_Peaks_Indexes"])
```

[image]

However, make sure to adjust correctly your sampling_rate (I set to 100 for illustrative purposes)

jaros1024, 2018-09-28:
Thanks! Do you think the error might be caused by wrong sampling_rate? To be honest, my data is so poorly documented that I won't be surprised if the sampling rate in documentation is wrong :)