neuropsychology/NeuroKit.py #79: missing recent releases (> 0.2.0)

boegel, 2018-10-02:
Recent versions are not being published on PyPI anymore (see https://pypi.org/project/neurokit/#history, most recent there is 0.2.0 which was added Sept 5th 2017), and new versions are not tagged here on GitHub either (see https://github.com/neuropsychology/NeuroKit.py/releases).

Based on current `master`, it seems like the latest version is 0.2.7 which corresponds to commit bba8ed9 (since that's where the version bump was done). But that doesn't include the bug fix from 8f166ad which was done shortly after...

Can you please i) tag releases on GitHub, ii) push releases to PyPI?

Letting people install straight from `master` without proper versioning results in a lot of confusion, which is easy to avoid by tagging releases.

DominiqueMakowski, 2018-10-02:
@boegel You're absolutely right, I'll try to do the requested updates ASAP

boegel, 2018-10-02:
@DominiqueMakowski Thank you very much!

Can you clarify which commit should be considered version 0.2.7?

DominiqueMakowski, 2018-10-02:
All commits done after the version bump should be considered 0.2.7 (in spite of the merging of a pre-0.2.7 branch issue)

boegel, 2018-10-02:
OK, so I'll go with current `master` (5dfa41d) as 0.2.7.