neuropsychology/NeuroKit.py #80: DLL load failed: The specified module could not be found.

vinhdh, 2018-10-03:
I used pip to install neurokit-0.2.7 followed by the installation guide on windows 10 x64 and python 3.7.0. The installation was successful but I encountered this error when importing the library as "import neurokit as nk".
I'm a newbie so I really appreciate every help to fix this.

> Exception has occurred: ImportError
DLL load failed: The specified module could not be found.
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 219, in _call_with_frames_removed
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap_external&gt;", line 728, in exec_module
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 677, in _load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 967, in _find_and_load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 983, in _find_and_load
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 219, in _call_with_frames_removed
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap_external&gt;", line 728, in exec_module
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 677, in _load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 967, in _find_and_load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 983, in _find_and_load
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 219, in _call_with_frames_removed
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap_external&gt;", line 728, in exec_module
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 677, in _load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 967, in _find_and_load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 983, in _find_and_load
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 219, in _call_with_frames_removed
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap_external&gt;", line 728, in exec_module
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 677, in _load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 967, in _find_and_load_unlocked
  File "C:\users\vinhd\desktop\py\&lt;frozen importlib._bootstrap&gt;", line 983, in _find_and_load
  File "C:\Users\vinhd\Desktop\py\hello.py", line 5, in <module>
    import neurokit as nk

DominiqueMakowski, 2018-10-08:
Hi @vinhdh, sorry for the delay I've been swamped at work. Well, could you provide some details, for example on "the installation was successful" (what makes you say that). Also, do other commands work properly? eg:

```python
import numpy as np

np.mean([1,2,3])
```

vinhdh, 2018-10-09:
Hi @DominiqueMakowski, I used VS Code on windows 10 to install neurokit by following command:
`pip install https://github.com/neuropsychology/NeuroKit.py/zipball/master`

This is the detail of the installation's progress:

> PS C:\Users\vinhd\Desktop\py> **pip install https://github.com/neuropsychology/NeuroKit.py/zipball/master**
> Collecting https://github.com/neuropsychology/NeuroKit.py/zipball/master
>   Downloading https://github.com/neuropsychology/NeuroKit.py/zipball/master
>      - 87.2MB 12.8MB/s
> Requirement already satisfied: numpy in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from neurokit==0.2.7) (1.15.2)
> Requirement already satisfied: pandas>=0.18.0 in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from neurokit==0.2.7) (0.23.4)
> Requirement already satisfied: scipy in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from neurokit==0.2.7) (1.1.0)
> Requirement already satisfied: sklearn in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from neurokit==0.2.7) (0.0)
> Requirement already satisfied: matplotlib in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from neurokit==0.2.7) (3.0.0)
> Requirement already satisfied: mne in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from neurokit==0.2.7) (0.16.2)
> Requirement already satisfied: bioread in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from neurokit==0.2.7) (1.0.4)
> Requirement already satisfied: nolds in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from neurokit==0.2.7) (0.4.1)
> Requirement already satisfied: biosppy in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from neurokit==0.2.7) (0.6.1)
> Requirement already satisfied: Pillow in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from neurokit==0.2.7) (5.3.0)
> Requirement already satisfied: cvxopt in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from neurokit==0.2.7) (1.2.1)
> Requirement already satisfied: python-dateutil>=2.5.0 in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from pandas>=0.18.0->neurokit==0.2.7) (2.7.3)
> Requirement already satisfied: pytz>=2011k in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from pandas>=0.18.0->neurokit==0.2.7) (2018.5)
> Requirement already satisfied: scikit-learn in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from sklearn->neurokit==0.2.7) (0.20.0)
> Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from matplotlib->neurokit==0.2.7) (2.2.2)
> Requirement already satisfied: cycler>=0.10 in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from matplotlib->neurokit==0.2.7) (0.10.0)
> Requirement already satisfied: kiwisolver>=1.0.1 in c:\users\vinhd\appdata\roaming\python\python36\site-packages (from matplotlib->neurokit==0.2.7) (1.0.1)
> Requirement already satisfied: setuptools in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from nolds->neurokit==0.2.7) (39.0.1)
> Requirement already satisfied: future in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from nolds->neurokit==0.2.7) (0.16.0)
> Requirement already satisfied: six in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from biosppy->neurokit==0.2.7) (1.11.0)
> Requirement already satisfied: shortuuid in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from biosppy->neurokit==0.2.7) (0.5.0)
> Requirement already satisfied: h5py in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from biosppy->neurokit==0.2.7) (2.8.0)
> Requirement already satisfied: bidict in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from biosppy->neurokit==0.2.7) (0.17.3)
> Requirement already satisfied: mkl in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from cvxopt->neurokit==0.2.7) (2019.0)
> Requirement already satisfied: intel-openmp in c:\users\vinhd\appdata\local\programs\python\python36\lib\site-packages (from mkl->cvxopt->neurokit==0.2.7) (2019.0)
> Installing collected packages: neurokit
>   Running setup.py install for neurokit ... done
> **Successfully installed neurokit-0.2.7**

[https://i.imgur.com/YJmqb09.png](url)

Everything's still ok until I added this line:
`import neurokit as nk` 

And I encountered the error as I mentioned
[https://i.imgur.com/DZUGjFx.png](url)

NachoCasta, 2018-11-05:
Hello, I have the same issue, is there a known fix or workaround?

DominiqueMakowski, 2018-11-05:
@vinhdh @NachoCasta to be honest I don't get it. If the package gets successfully installed, it should successfully get imported. @NachoCasta are you also using visual studio?

DominiqueMakowski, 2018-11-06:
@NachoCasta Awesome, so neurokit worked then?

NachoCasta, 2018-11-06:
Yes! The problem was cvxopt which needs numpy+mkl to work.

Ignacio Castañeda

DominiqueMakowski, 2018-11-06:
Awesome! I'll close the issue then 😄 Thanks a lot @NachoCasta for digging into it and reporting the solution!

closed 2018-11-06 by DominiqueMakowski

vinhdh, 2018-11-07:
Installing numpy+mkl fixed this error. Now it works very well. Thank you @NachoCasta