neuropsychology/NeuroKit.py #84: Allow custom markers in nk.plot_events_in_signal function

renatosc, 2019-01-09:
Right now, function [nk.plot_events_in_signal](https://neurokit.readthedocs.io/en/latest/_modules/neurokit/signal/events.html#plot_events_in_signal) always uses a vertical line to show the events in the signal. That sometimes makes it difficult to precisely see the events position in the signal. 

Allowing us to specify custom markers (in the same way that we can now specify custom colors) would provide us with better graphics.

DominiqueMakowski, 2019-01-10:
@renatosc Thanks for the suggestion, do you have any idea about what types of markers you'd like to see?

renatosc, 2019-01-10:
https://github.com/neuropsychology/NeuroKit.py/pull/85

DominiqueMakowski, 2019-01-12:
All merged ;) thanks again

closed 2019-01-12 by DominiqueMakowski