neuropsychology/NeuroKit.py #92: Peaks information in cardiac cycles

renatosc, 2019-03-11:
Hi Dominique,

The ecg_process today returns the peaks (P,Q, R, ...) in reference to the sample number, so I can easily plot both the entire ECG and the peaks.   Since the Cardiac Cycles returned by ecg_process starts at the same point (zero, making it possible to have that plot of all cycles on top of each other), the peaks info returned cannot be used directly.  I was wondering if there is any built-in feature to show the peaks in reference to the Cardiac Cycles. (i.e, I would like get the peaks position inside each cardiac cycle array entry).

DominiqueMakowski, 2019-03-12:
@renatosc Yes, I would say that this would work. However, an even simpler workaround would be to subtract from each peak from the signal the index of the corresponding R peak. This way, all of the peaks would be in reference to *their* R peak, and you could quite easily plot them all together?

Or indeed, you could also re-scale these deltas and add them in the cardiac_cycle data? But the general idea would be to use the R peaks indices as a reference for the other values.

renatosc, 2019-03-12:
@DominiqueMakowski , thanks for the feedback. I ended up just using the delta in order to avoid having to deal with negative values. I will see later how can I added that to NK and make a pull request.

closed 2019-03-12 by renatosc