neuropsychology/NeuroKit.py #93: Python 2.7: TypeError: unbound method materials() must be called with Path instance as first argument (got nothing instead)

MarcoFilippi, 2019-03-12:
`import neurokit as nk
import pandas as pd
import numpy as np
import seaborn as sns

df = pd.read_csv("https://raw.githubusercontent.com/neuropsychology/NeuroKit.py/master/examples/Bio/bio_100Hz.csv")
bio = nk.bio_process(ecg=df["ECG"], sampling_rate=100)
nk.z_score(bio["df"]).plot()`

Trying to run the template I got the following error message:

> Traceback (most recent call last):
  File "neuroTest.py", line 10, in <module>
    bio = nk.bio_process(ecg=df["ECG"], sampling_rate=100)
  File "/home/sisu/anaconda2/lib/python2.7/site-packages/neurokit/bio/bio_meta.py", line 125, in bio_process
    ecg = ecg_process(ecg=ecg, rsp=rsp, sampling_rate=sampling_rate, filter_type=ecg_filter_type, filter_band=ecg_filter_band, filter_frequency=ecg_filter_frequency, segmenter=ecg_segmenter, quality_model=ecg_quality_model, hrv_features=ecg_hrv_features, age=age, sex=sex, position=position)
  File "/home/sisu/anaconda2/lib/python2.7/site-packages/neurokit/bio/bio_ecg.py", line 119, in ecg_process
    quality = ecg_signal_quality(cardiac_cycles=processed_ecg["ECG"]["Cardiac_Cycles"], sampling_rate=sampling_rate, rpeaks=processed_ecg["ECG"]["R_Peaks"], quality_model=quality_model)
  File "/home/sisu/anaconda2/lib/python2.7/site-packages/neurokit/bio/bio_ecg.py", line 357, in ecg_signal_quality
    model = sklearn.externals.joblib.load(Path.materials() + 'heartbeat_classification.model')
TypeError: unbound method materials() must be called with Path instance as first argument (got nothing instead)

DominiqueMakowski, 2019-03-12:
@MarcoFilippi, it seems like this issue is created when neurokit is trying to load the sklearn model (saved using sklearn for python 3). I am not sure that I can fix it, as it seems tied to scikitlearn 😕  Maybe try updating your version of scikit-learn?

The goal is being able to load [this file](https://github.com/neuropsychology/NeuroKit.py/tree/master/neurokit/materials)... Does it work if you try only that? 
`model = sklearn.externals.joblib.load('heartbeat_classification.model')`

MarcoFilippi, 2019-03-12:
@DominiqueMakowski , I updated to scikit-learn-0.20.3 and I tried to load the file with the command you suggested! Now the message I received is the following:

> Traceback (most recent call last):
  File "classes.py", line 3, in <module>
    model = sklearn.externals.joblib.load('./heartbeat_classification.model')
  File "/home/sisu/anaconda2/lib/python2.7/site-packages/sklearn/externals/joblib/numpy_pickle.py", line 598, in load
    obj = _unpickle(fobj, filename, mmap_mode)
  File "/home/sisu/anaconda2/lib/python2.7/site-packages/sklearn/externals/joblib/numpy_pickle.py", line 526, in _unpickle
    obj = unpickler.load()
  File "/home/sisu/anaconda2/lib/python2.7/pickle.py", line 864, in load
    dispatch[key](self)
  File "/home/sisu/anaconda2/lib/python2.7/pickle.py", line 892, in load_proto
    raise ValueError, "unsupported pickle protocol: %d" % proto
ValueError: unsupported pickle protocol: 3

Which is your scikitlearn version? I suppose I cannot avoid to jump to python3, since scikit-learn website says "Scikit-learn 0.21 will require Python 3.5 or newer."