neuropsychology/NeuroKit.py #96: TypeError: 'str' object is not callable

anandketki95, 2019-06-01:
I am getting type error while using bio_process. 
Below is my code. 
"i = 1
bio = nk.bio_process(ecg=df_respiban[i][:]['ECG'],rsp=df_respiban[i][:]['RESP'], ecg_quality_model=None, ecg_filter_frequency=[1,35], sampling_rate=100)"

Error is as follows,

~\Anaconda3\lib\site-packages\nolds\measures.py in sampen(data, emb_dim, tolerance, dist, debug_plot, debug_data, plot_file)
    762     # successively calculate distances between each pair of template vectors
    763     for i in range(len(tVecsM) - 1):
--> 764       dsts = dist(tVecsM[i + 1:], tVecsM[i])
    765       if debug_plot:
    766         plot_data[-1].extend(dsts)

TypeError: 'str' object is not callable

[Untitled9.pdf](https://github.com/neuropsychology/NeuroKit.py/files/3244071/Untitled9.pdf)

DominiqueMakowski, 2019-06-03:
Hi @anandketki95, could you provide or send me your data or a sample of it so I can try reproducing the error? Thanks!

aravali324, 2019-07-08:
Hello , When i run the default  example provided in documentation,  i am facing the same problem

matrach, 2019-07-15:
Hi,

the issue is because nolds deprecated passing ``dist`` as string to ``measures.sampen`` and removed it in `0.5.0`.
The issue is located [here](https://github.com/neuropsychology/NeuroKit.py/blob/master/neurokit/signal/complexity.py#L145]). Just pass ``nolds.measures.rowwise_chebyshev`` instead.

As a quickfix, one may install an old version of nolds:
``pip install 'nolds<0.5'``