neuropsychology/NeuroKit.py #97: I can't see my trigger channel on Neurokit

ArnaudSteinmetz, 2019-06-25:
Hi, i'm both new in python and data processing.
I have a EDA file from Biopac ( acq). And when i open it with neurokit it displays only 2 channels:
-one is the  EDA response
-the second is the digital input but equal to 0 all along.
The thing is, it's supposed to have 8 triggers.
When i try read it with bioread, the 8 triggers are detected.
Can someone know why?

Regards,

Arnaud

DominiqueMakowski, 2019-06-25:
Hi @ArnaudSteinmetz, welcome to github ☺️ 

Hum that's weird, because the function to read acknowledge file in neurokit uses bioread. During recording, do these channels actually show on their digital channel (like `____|---|____|----|___`)?

Worst case scenario, if you manage to retrieve the timings from bioread, you can still add them a posteriori

ArnaudSteinmetz, 2019-06-26:
hi, the thing is i'm in intership and i didn't do the recording.
But when i looked at the data on biopac i saw the triggers. I don't remember the exact shape they had.

But i want to proceed anyway is that hard to get the timings from bioread? 
i recontact you when i know more about that recording ( it may be in some time i'll be away for a few days)

DominiqueMakowski, 2019-06-27:
Could you provide me with some example file and a reproducible code so I can reproduce the issue? Thanks!

ArnaudSteinmetz, 2019-06-28:
Hi, @DominiqueMakowski ,

Here is the figure i get with pandas after the extraction with neurokit.

[image]
[Sujet12_PasT.zip](https://github.com/neuropsychology/NeuroKit.py/files/3338149/Sujet12_PasT.zip)

But when i read it with bioread, it gives me this
"AcqKnowledge file (rev 128): 9 channels, 2000.0 samples/sec".

As you can see, Bioread detects 9 trigger channels (which is sipposed to be, EDA and 8 trigger channels).
But on the graph, NK detectcs only 2= EDA and one digital input (equal to 0 all along).

I give you the few lines i used to read this and the file i used to tru that.

`import mne
import neurokit as nk
import bioread
import matplotlib.pyplot as plt
import pandas as pd
import numpy as np
#%%

df,SR=nk.read_acqknowledge(r"C:\DATA\EDA\Sujet12_PasT",return_sampling_rate=True)

S12=bioread.read_file(r"C:\DATA\EDA\Sujet12_PasT.acq")`

Regards

iPsych, 2023-02-10:
@ArnaudSteinmetz 
I hope you found solution ago.
That's because neurokit cannot detect the duplicate channel names 'Digital Input' of .acq file.
Acqknowledge put the same default names for all digital channels by default.
Once you change the separate digital channels to different names (i.e. 'Digital Input 1', 'Digital Input 2'...), You will see the nk.read_acqknowledge function works.

Actually, bioread was impressive to handle duplicate name flawlessly.