neuropsychology/NeuroKit.py #105: Bio_process error with sampling rate of 2000

femicarrington, 2020-01-18:
Hi there,

I am using Neurokit to analyze EDA data for the first time and I'm having an issue with the bio_process function.  My data was acquired with 2000 samples/sec. When I set the sampling rate to 1000, there is no issue, however, when I set the sampling rate to 2000 samples/sec I get an error: 
ValueError: operands could not be broadcast together with shapes (20704,) (20627,).

Here is the code I'm using:
`bio_features = nk.bio_process(eda=df["CH10"], emg=df["CH1"], add=df_markers, sampling_rate=2000, scr_method='makowski', scr_treshold=0.1)

nk.z_score(bio_features["df"]).plot()
plt.ylim(-4,4)
plt.show`

Thanks in advance for your help!

DominiqueMakowski, 2020-01-23:
Hei @femicarrington Sorry for the late answer! As you know, we are currently re-implementing, fixing and improving a lot of NeuroKit's content in the so-called [neurokit2](https://github.com/neuropsychology/NeuroKit), which ultimately aims at to replace neurokit1. 

Could you try installing the neurokit2 from Github and run [`eda_process`](https://github.com/neuropsychology/NeuroKit#electrodermal-activity-eda-processing) to see if at least this works? Thanks!

EstherJohanna, 2020-01-27:
Hi Dominique,

I had exactly the same problem and eda_process in neurokit2 works also with a sampling rate over 1000!
:)

DominiqueMakowski, 2020-01-27:
@EstherJohanna glad to hear that! We are in the process of re-implementing most ofthe features from nk1 to nk2, so stay tuned :)