neuropsychology/NeuroKit.py #106: How to compute QRS duration time period??

advent-embedded, 2020-02-13:
I could able to compute the indices of P, Q R, i need the duration of QRS, to arrive this i need S index of the wave. i there a in-built routine which does this

advent-embedded, 2020-02-13:
with version 0.2.8 i get empty list for all peak and onset Values, with 0.2.0 i get only peak values for T, P & Q. is ter any fix?

DominiqueMakowski, 2020-02-14:
@advent-embedded While I cannot really say for neurokit1, as the QRS delineation algorithm is quite weak. That being said, we are currently in the process of implementing a new and more robust algorithm for delineation in [neurokit2](https://github.com/neuropsychology/NeuroKit), which should be ready in a few days (hopefully @Tam-Pham). 

I will transfer this issue over to NK2 and will update you when it is done!